gut-space/svarog-server #33: Fix TCA values

tomaszmrugalski, 2021-02-19:
Right now the station uploads TCA values equal to AOS. This is clearly incorrect. There are two tasks to do here:

1. correct the upload script
2. correct existing entries in the DB

tomaszmrugalski, 2021-02-20:
On a second thought, maybe a better approach would be to get rid of TCA completely? Since the orbits we're dealing with are nearly circular, we can use basic formula to calculate it `TCA = (AOS + LOS)/2`.

fivitti, 2021-02-20:
It may be not right that TCA is average of LOS and AOS.
In our context AOS and LOS means start and stop recording. 
We use a strategy which slice of pass should be record. TCA may be not recorded.

For example: NOAA 15 has real AOS at 12:00 and LOS 13:00. TCA is 12:30.
But we also have a pass of Meteor with real AOS 12:20 and LOS 13:00 (TCA 12:40).
Meteor has better quality pass (centrally over station).
Then strategy select for record:

- NOAA 15 from 12:00 to 12:20
- Meteor from 12:20 to 13:00

We insert these times as AOS and LOS. In this case the closest point to station (TCA) of NOAA 15 will be at 12:20.

tomaszmrugalski, 2021-05-15:
The observation submission is now improved, see commit 59e2c55224de7b702c704f4f25450b30618d6b50 on #45. It's not a proper TCA, just an average between start of observation and LOS.