gut-space/svarog-server #34: python 3.9

tomaszmrugalski, 2021-02-20:
The PR #32 removed python 3.6. We need to keep up with the progress and enable 3.9 some time soon. The [first attempt](https://travis-ci.com/github/gut-space/aquarius/jobs/484711249) was a failure, due to unavailable numba in expected version.

fivitti, 2021-02-20:
We must to wait. [Source from 11 January](https://github.com/numba/llvmlite/issues/531#issuecomment-757782846):

> The Numba/llvmlite stack currently does not have a release to support Python 3.9. For llvmlite this means there is no binary wheel for Python 3.9 and an attempt to install llvmlite from PyPi will result in an attempted compilation of llvmlite which in turn yields the error you are observing.
>
> There is an issue tracking this here: numba/numba#6579 and we are hoping to release Python 3.9 support soon (where soon == couple of weeks).

tomaszmrugalski, 2021-11-18:
This has been addressed with #62.

closed 2021-11-18 by tomaszmrugalski