gut-space/svarog-server #39: SoapySDR integration

fivitti, 2021-02-27:
We have new SDR! But it isn't compatible with ```librtlsdr``` tools. I think that our station should be independent on hardware.

We can use [SoapySDR](https://github.com/pothosware/SoapySDR) as layer between hardware and our software. It provides plugins for support many of SDRs.

I notice that PyPI has a Python-wrapper for it: [SimpleSoapy](https://pypi.org/project/SimpleSoapy/).
I think that we can use it in Python-based recipes.

fivitti, 2021-02-27:
SoapySDR has own [Python binding](https://github.com/pothosware/SoapySDR/wiki/PythonSupport).

fivitti, 2021-02-27:
I installed additional packages:

```
apt install soapysdr-tools python3-soapysdr
```

fivitti, 2021-02-27:
We decided to use ```rx-tools```.

For it I installed first: ```libsoapysdr-dev``` and next build the tools using [official instruction](https://github.com/rxseger/rx_tools).
At end I also call ```make install```.

tomaszmrugalski, 2021-05-09:
SoaPy seems to be a good choice. This is what satnogs is using. See https://wiki.satnogs.org/SatNOGS_Client_Setup.