gut-space/svarog-server #55: Fix rating calculation and upload

tomaszmrugalski, 2021-06-11:
I'd like to be able to compare the performance of noaa-apt and gnu-radio recipes against each other. To do that, image quality assessment is necessary. I looked at the analog rating protocol, enabled it in the config, but it didn't help. I've checked the code and pushed a fix in 31de664466cc1f0e606e1a8a5be3e24969e910ea. Before that fix, the rating was always `None`. 

This now uncovered two problems:
- [ ] The station calculates score of 1.0 even for poor images, e.g. the station calculated 1.0 for [observation#3881](https://svarog.space/obs/3881).
- [ ] Something was broken in the score upload to the server. Station logs say this:

```
receiver.py:88: Getting rating for product /home/pi/observations/2021-06-11-1904-noaa-18/product.png (rating algorithm is analog)
receiver.py:90: Product /home/pi/observations/2021-06-11-1904-noaa-18/product.png got rating 1.0
```

yet the server shows no rating. Need to debug it.

tomaszmrugalski, 2021-06-11:
I've looked around a bit and I think there's a couple things that could be improved:

1. The notes field is currently unusable. Hardcoded to `` in one place and to `...` in another. This should read a content of some file, e.g. `~svarog/notes.txt`. The idea behind this is that the station maintainer could put there some info easily, e.g. `running on airspy now`, `disabled LNA` etc. This kind of things would be easy to keep up to date and there would be a written trace of the changes done over time.
2. The log file is not uploaded. It could be another product.
3. I strongly feel that we need to introduce some structured data here that would convey the parameters, such as frequency, Doppler correction, name of the recipe, rating method etc. The syntax should be extensible, so adding more parameters (e.g. all those knobs in gnu radio) could be transparent and wouldn't require server changes. If done correctly, this could easily replace the single value rating field.

fivitti, 2021-06-12:
> I strongly feel that we need to introduce some structured data here that would convey the parameters, such as frequency, Doppler correction, name of the recipe, rating method etc. The syntax should [...]

It may be useful to assign any key-value dictionary for observations. Data in the dictionary may contain any attributes, variety for each antenna, station, satellite, etc.

Postgres has some possibilities to handle this case. The first is JSON data type. It may contain any data and Postgres has few functions to query the JSON content. But JSON data type isn't intended for this (in my opinion). It is very good for store JSON as a single blob, but functions to operate on it are just an addition.
Probably better approach will be to use "hstore" official Postgres extensions to support key-value columns. It should be a little more efficient to query them for example for filtering.

We should analyze if key-value data should be assigned to observation (as "notes") or to a specific product.

tomaszmrugalski, 2021-06-12:
I was thinking about JSON, but can do some experiments with `hstore`. I know there are queries possible that look into JSON keys.

For the time being I think it's ok to have the meta-data (or key-value or whatever you want to call it) to be associated with the whole observation. This will be useful as it is. If we need to get it per product some time later, we can always add additional meta-data for products. This may be useful later, e.g. when we'll start receiving SSTV images from ISS and it would transmit multiple images during one pass. Each image could have its own meta-data (width, height, quality, etc.).

tomaszmrugalski, 2021-06-12:
I've updated the station code and it [calculated the rating properly](https://svarog.space/obs/3896)!

This indeed fixes the problem.

The score is unreasonably high (41%), but there's a good explanation. Gnu Radio recipe always produces an image of constant height, with the remainder part being black. The rating algorithm treats black parts are noiseless, and thus good.
That it, however, something I can fix in the recipe. The algorithm now works fine. As such, closing this.

closed 2021-06-12 by tomaszmrugalski