gut-space/svarog-server #57: Argparse for submitobs.py

fivitti, 2021-06-12:
Script "submitobs.py" has a lot of parameters to pass from the command line. They are parsed directly from ```sys.argv```. The order of parameters is important and some of them are optional. 
I think that using ```argparse``` library to handle them will be more user-friendly and descriptive.

tomaszmrugalski, 2021-06-12:
Sounds good to me. This script wasn't used manually before.

I was thinking that one important change in this area would be to allow multiple files. At the moment at least 3 come to mind: products, waterfall, log file. One way to do it would to allow coma separated list of files as the first argument.

tomaszmrugalski, 2021-06-12:
Yesterday, I've ran my local server (`cd server; ./svarog-web.py`) and then edited the station config to point to `'url': 'http://localhost:8080/receive'`. Worked nicely, except I couldn't make the signatures to work. Most likely I messed up the secret configuration.